Tombraider2006/klipperFB6 #5: UART cable solder

ColdBeer72, 2023-04-06:
Hi!
In your page you say it's obsolete to desolder the resistors, but pictures show there is no resistor yet in the board...
I suppose those are pictures with older method used.
Do you mean I could directly solder the cable right in the upper side of the resistors (towards the drivers)?

Tombraider2006, 2023-04-06:
you god damm right. on pictures old method.

ColdBeer72, 2023-04-06:
Oh thanks!
So, it's really secure to solder the cable directly in the resistor (driver's side), isn't it?

Tombraider2006, 2023-04-06:
yes, this has already been tested by several users, everything works. it also eliminates the risk of breaking the track when soldering

Tombraider2006, 2023-04-06:
be sure to use hot melt adhesive to avoid breaking the wire from vibration during future operation of the printer.

Tombraider2006, 2023-04-06:
green side of course.

ColdBeer72, 2023-04-06:
Thanks again for the clarification

closed 2023-04-06 by ColdBeer72